solokeys/solo1-cli #2: `solo ls` fails on Windows user install

ignaloidas, 2019-02-18:
After installing solo-python on Windows 10 with `pip install --user solo-python` trying to run `solo ls` fails with this output
```
C:\Users\User>solo ls
:: Solos
b'\\\\?\\hid#vid_0483&pid_a2ca#6&2c612bc6&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\x00': Solo HACKER (Unlocked)
Traceback (most recent call last):
  File "c:\program files\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\program files\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\User\AppData\Roaming\Python\Python37\Scripts\solo.exe\__main__.py", line 9, in <module>
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\solo\cli\__init__.py", line 112, in ls
    st_dfus = solo.dfu.find_all()
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\solo\dfu.py", line 52, in find_all
    st_dfus = usb.core.find(idVendor=0x0483, idProduct=0xDF11, find_all=True)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\usb\core.py", line 1263, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```
At this time I have a solo hacker inserted and 'solo key wink`, `solo key verify`  and `solo key rng` commands work.

conorpp, 2019-02-18:
I forgot about this issue.  To access the ST DFU, Windows users need to install [libusb](https://github.com/libusb/libusb/wiki/Windows).  But more most things, the ST DFU isn't used.

Accessing any solo features or running a normal update, all uses the HID interface which doesn't need an extra driver.

For `solo ls`, it prints the Solo devices correctly, then crashes upon looking for ST-DFU devices.  I recommend we simply catch this error and print a more helpful message.

nickray, 2019-02-18:
Hopefully this fix in v0.0.5 helps: https://github.com/solokeys/solo-python/blob/6915722f9bd8a25f3eab982cbcc1544666bc10e5/solo/cli/__init__.py#L121

Maybe `solo ls` should not even look for ST DFUs unless passed an option like `--dfus-too`.

nickray, 2019-02-19:
I changed `solo ls` behaviour in v0.0.6a1 (get it with `pip install -U --pre solo-python`) to not even search for ST DFUs except with `solo ls --all`.

ignaloidas, 2019-02-19:
Works fine now, closing

closed 2019-02-19 by ignaloidas