solokeys/solo1-cli #3: `solo key update` fails on Windows

ignaloidas, 2019-02-18:
`solo key update --hacker` fails with this message:
``` 
C:\Users\User>solo key update --hacker
Traceback (most recent call last):
  File "c:\program files\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\program files\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\User\AppData\Roaming\Python\Python37\Scripts\solo.exe\__main__.py", line 9, in <module>
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\solo\cli\update.py", line 63, in update
    solo_client.is_solo_bootloader()
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\solo\client.py", line 243, in is_solo_bootloader
    self.bootloader_version()
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\solo\client.py", line 185, in bootloader_version
    data = self.exchange(SoloBootloader.version)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\solo\client.py", line 155, in exchange_u2f
    res = self.ctap1.authenticate(chal, appid, req)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\fido2\ctap1.py", line 259, in authenticate
    response = self.send_apdu(ins=CTAP1.INS.AUTHENTICATE, p1=p1, data=data)
  File "C:\Users\User\AppData\Roaming\Python\Python37\site-packages\fido2\ctap1.py", line 223, in send_apdu
    raise ApduError(status, data)
fido2.ctap1.ApduError: (27264, b'')
```

nickray, 2019-02-18:
This should fail in a more helpful way now:
https://github.com/solokeys/solo-python/blob/6915722f9bd8a25f3eab982cbcc1544666bc10e5/solo/cli/update.py#L83-L86
Please update to v0.0.5

conorpp, 2019-02-18:
Running on Windows, I get this error when I try to update an old key as well.  Only when it is not in bootloader mode though.  If I hold the button down, plug it in, and wait for the flashing yellow LED first, then the update process works.

Was this from bootloader mode or "normal" mode?

ignaloidas, 2019-02-19:
I wasn't in bootloader mode. When in bootloader everything works fine.

closed 2019-02-19 by ignaloidas