solokeys/solo1-cli #9: Invalid parser option

szszszsz, 2019-03-06:
Solo crashes just after running. It looks like one of the parser options are invalid. Log below.

Version: `solo-python==0.0.6`
Python: Python 3.7.2
OS: Fedora 29

Installed without errors via ` pip3 install solo-python --user`.

```
$ solo
Traceback (most recent call last):
  File "/home/sz/.local/bin/solo", line 7, in <module>
    from solo.cli import solo_cli
  File "/home/sz/.local/lib/python3.7/site-packages/solo/cli/__init__.py", line 17, in <module>
    from solo.cli.key import key
  File "/home/sz/.local/lib/python3.7/site-packages/solo/cli/key.py", line 18, in <module>
    from solo.cli.update import update
  File "/home/sz/.local/lib/python3.7/site-packages/solo/cli/update.py", line 41, in <module>
    help="Development option: pull firmware from http://localhost:8000",
  File "/usr/lib/python3.7/site-packages/click/decorators.py", line 170, in decorator
    _param_memo(f, OptionClass(param_decls, **attrs))
  File "/usr/lib/python3.7/site-packages/click/core.py", line 1459, in __init__
    Parameter.__init__(self, param_decls, type=type, **attrs)
TypeError: __init__() got an unexpected keyword argument 'hidden'
```

nickray, 2019-03-06:
Are you using an old version of click? Current is 7.0, should have `hidden`: https://github.com/pallets/click/pull/500

szszszsz, 2019-03-07:
Hi!

Yes, I do apparently:
```
$ pip3 freeze | grep click
click==6.7                                                            
click-completion==0.3.1                                               
```
Update with `pip3 install --upgrade click --user` helped, thank you! In that case adding a hard version dependency would prevent this issue to occur on other installations.

nickray, 2019-03-08:
Fixed in v0.0.7

closed 2019-03-18 by nickray