solokeys/solo1-cli #12: No matching distribution found for solo-python

LoungeFlyZ, 2019-03-08:
I'm a noob and getting the following on Mac OS (10.14.1) when trying to install via pip3:

```
>pip3 install solo-python
Collecting solo-python
  Could not find a version that satisfies the requirement solo-python (from versions: )
No matching distribution found for solo-python
```

Any pointers?

nickray, 2019-03-09:
does `python3 --version` return at least 3.6?

LoungeFlyZ, 2019-03-09:
hmmm something weird going on... 

```
>python3 --version
Python 3.5.1

> python --version
Python 3.7.2
```

nickray, 2019-03-18:
@LoungeFlyZ If this is an issue on your side can you close this please? If it is an issue on our side, please clarify.

LoungeFlyZ, 2019-03-22:
looks to be a problem with python install on my machine. closing.

closed 2019-03-22 by LoungeFlyZ