solokeys/solo1-cli #20: Add PIN support

nickray, 2019-04-08:
There are two parts:

- allow passing `-p/--pin <PIN>` in all subcommands that might need it
- implement subcommands `solo pin set/reset` to set/change the PIN

There is working code in https://github.com/solokeys/solo/blob/master/tools/testing/tests/fido2.py to guide this.

The actual code should be added to [client.py](https://github.com/solokeys/solo-python/blob/master/solo/client.py), which the CLI just calls.

szszszsz, 2019-06-07:
Hi!
I would argue, whether users should be encouraged to provide PIN in the command line. Perhaps support the environment variable instead?

nickray, 2019-08-07:
I agree it would be nice to (more generally) have a framework to read `solo` CLI arguments out of environment variables. Maybe this is a separate task though?

Not sure how much more secure this is, won't most people type `export SOLO_PIN=xxx`? :)

szszszsz, 2019-08-08:
Sounds like a generic thing to do. I remember some CLI frameworks have such feature available - hasn't `click` support that already?

I think the idea back then was to avoid cluttering the shell's history with the raw PIN, which is written to user's hard drive in plaintext. I agree, that using `export` or `env` while executing command defeats that.

cuongnv, 2019-10-30:
I add change pin feature to solo tool
https://github.com/solokeys/solo-python/pull/42

My1, 2019-11-09:
might it not just be better to ask the PIN inside the running command? similar like ssh-keygen or whatever it was called does?

because env vars are semi annoying to use and most importantly annoying when dealing with multiple keys when you have different PINs, also browsers and stuff probably have a reason why they always ask for the pin and not store it somewhere...