solokeys/solo1-cli #22: Running as root in Docker

conorpp, 2019-05-13:
When making a bundled firmware in Docker, it runs into this issue.

```
+ /opt/conda/bin/solo mergehex bootloader-nonverifying-2.3.0.hex firmware-hacker-2.3.0.hex bundle-hacker-2.3.0.hex
THIS COMMAND SHOULD NOT BE RUN AS ROOT!

Please install udev rules and run `solo` as regular user (without sudo).
We suggest using: https://github.com/solokeys/solo/blob/master/udev/70-solokeys-access.rules

For more information, see: https://docs.solokeys.io/solo/udev/
```

Maybe turn the root message into a warning?

nickray, 2019-05-23:
That, or there's an existing mechanism to override the warning: `sudo ALLOW_ROOT= /path/to/solo`. What do you think about adding this info to the (aborting) warning, and also to the Docker build script?

I'd also be fine with just outputting a warning and continuing.

conorpp, 2019-05-27:
I think the warning is easiest.  I think the only reason we put the sudo restriction in there was to catch udev issues, and I think a warning would still catch it fine.

closed 2019-05-27 by nickray

nickray, 2019-05-27:
Fixed in https://github.com/solokeys/solo-python/releases/tag/0.0.11