solokeys/solo1-cli #24: Win 10 1903 USB Issue

onlykey, 2019-05-22:
Have you guys tested this on the new version of Windows 10 that is getting pushed out 1903? It looks like with this release solo-python only works if run as administrator. Seems to be USB HID related as it is affecting other USB HID devices I have tested as well. To force Windows to update to 1903 to test this you can opt in to the Windows insider program. 

[image]

conorpp, 2019-05-22:
Yeah, I've noticed this.  I think accessing FIDO2 keys via windows API is okay for users, but using direct USB API is admin only.  I'm not sure how best to fix other than using an admin prompt.

nickray, 2019-05-24:
Looks like it's PC/SC all over again: "Linux should do it our way too": https://groups.google.com/a/fidoalliance.org/forum/m/#!topic/fido-dev/iDLKY-HsY0A

nickray, 2020-01-12:
Confirm that Windows locked down access. Not in our hands.

PR #45 (2020-02-27): Refactoring the init warnings
This PR resolves #24 by adding warning on Windows 10 1903 and higher for the requirement of administrative privileges. I think that it is a good idea and avoids user confusion - at least until the fido2 library or solo gets proper workaround.

[image]

Together with the Windows warning, I did refactoring for the Linux root warning as well.
closed 2020-01-12 by nickray