solokeys/solo1-cli #26: solo key verify crashes instantly

Vash63, 2019-08-15:
I've tested this on both a macos and archlinux system, both running python 3.7.4, and in both cases `solo key verify` causes an immediate crash. 

```
$ solo key verify
Please press the button on your Solo key
Traceback (most recent call last):
  File "/usr/bin/solo", line 10, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.7/site-packages/solo/cli/key.py", line 219, in verify
    cert = solo.client.find(serial).make_credential()
  File "/usr/lib/python3.7/site-packages/solo/client.py", line 215, in make_credential
    attest, data = self.client.make_credential(rp, user, challenge, exclude_list=[])
  File "/usr/lib/python3.7/site-packages/fido2/client.py", line 308, in make_credential
    pin, timeout, on_keepalive
  File "/usr/lib/python3.7/site-packages/fido2/client.py", line 324, in _ctap2_make_credential
    raise ValueError('PIN required!')
ValueError: PIN required!
```

It crashes within a split second of me hitting enter before I have a chance to try and press the button on the key. My key is a Solo Secure running 2.4.2.

My1, 2019-08-16:
it does work when the solo has no pin set however. if you have no important accounts on your solo, you can reset then verify

nickray, 2019-08-16:
Related: https://github.com/solokeys/solo-python/issues/20.

I will have to check (it's been a while) whether it's possible to use simultaneously "direct attestation" (since we want to "verify" the key) and no user verification. Is this what a "discouraged" UV does? In this particular case, would be helpful for sure.

My1, 2019-08-17:
user verification has 3 levels.
preferred (iirc default):vverify the user if you can. if you cant (e.g. no pin set), no problem
required: force user verification
discouraged: do NOT use verification

nickray, 2019-08-19:
@My1 this is not quite true. See https://fidoalliance.org/specs/fido-v2.0-rd-20170927/fido-client-to-authenticator-protocol-v2.0-rd-20170927.html#authenticatorMakeCredential "If pinAuth parameter is not present and clientPin been set on the authenticator, return CTAP2_ERR_PIN_REQUIRED error."

nickray, 2019-08-19:
@Vash63: I release a new version 0.0.13. Can you test by running `pip3 install -U --user solo-python` (or whatever update method makes sense in your setup), and then `solo key verify --pin <your-PIN>`?

My1, 2019-08-19:
https://github.com/solokeys/solo-python/issues/26#issuecomment-522716534 does makecredential only create a resident credential or also a normal one? because normal credentials can apparently be generated without pin like for example on the yubico demo site with both a yubi5 and a solo which both have a pin set.

Vash63, 2019-08-22:
@nickray It's working now, thanks.

nickray, 2019-08-22:
Yes, right now `click` just passes through most errors as traceback. Pull requests welcome!

closed 2019-08-22 by nickray