solokeys/solo1-cli #31: Unexpected 0x41 markers added during merge

szszszsz, 2019-08-24:
Hi!

While working on https://github.com/solokeys/solo/pull/238 I have noticed a `0x41` marker on the `APPLICATION_END_PAGE` boundary, introduced during the `all.hex` merge. Is it needed anywhere, or is it just for debug purposes (e.g. to hand-check firmware dump)?

https://github.com/solokeys/solo-python/blob/e2a81a2c39bef61f98dca29154c5fdedff3a1044/solo/operations.py#L68-L69

conorpp, 2019-08-24:
I believe those were there originally for debugging purposes yes.  I can't find or remember any other reason.

szszszsz, 2019-09-06:
OK! Let them be. Perhaps it would be worth making a comment to the setting lines in `solo-python` about their purpose. This will make the ticket actionable.

nickray, 2020-01-12:
Closing for inactivity.

closed 2020-01-12 by nickray