solokeys/solo1-cli #34: rng doesn't check RNG_SR

NicolaiSoeborg, 2019-08-30:
It seems like the `get_rng` command doesn't check the "RNG status register"  (`RNG_SR`).

From page 659 of the reference manual [0]:
> When a noise source (or seed) error occurs, the RNG stops generating random numbers and sets to “1” both SEIS and SECS bits to indicate that a seed error occurred. If a value is available in the RNG_DR register, it must not be used as it may not have enough entropy.

The check seems to be done in the embedded code at:  https://github.com/solokeys/solo/blob/master/targets/stm32l432/src/rng.c#L28  but why is the check and'ing with `0x66` ?  The documentation says that CEIS and SEIS are bit 5 and 6, so shouldn't the check be `RNG->SR & 0x60` or `RNG->SR & 0x06` depending on endianness?).

[0] RM0394:  https://www.st.com/content/ccc/resource/technical/document/reference_manual/group0/b0/ac/3e/8f/6d/21/47/af/DM00151940/files/DM00151940.pdf/jcr:content/translations/en.DM00151940.pdf

conorpp, 2019-08-30:
It's only checking one byte, so endianness isn't an issue.  `0x60` checks the CEIS and SEIS bits (bits 5, 6), and the `0x06` checks SECS and CECS (bits 1, 2).

I suppose `RNG->SR & 0x60` would be sufficient, since they indicate that either SECS or CECS would be set.  But might as well check all at once (`RNG->SR & 0x66`).

nickray, 2020-01-12:
Wrong repo.

closed 2020-01-12 by nickray