solokeys/solo1-cli #38: Can't update in a clean Ubuntu (18.04) installation

johnnyasantoss, 2019-10-24:
I'm trying to update my solo key on a new laptop (with udev rules already installed) but it is failing.
```log
solo key update --secure
We are about to update with the latest Solo Secure firmware.
Please confirm that the connected Solo key is a Solo Secure [y/N]: y
Wrote temporary copy of firmware-secure-2.5.3.json to /tmp/tmpgvas76y9.json
sha256sums coincide: 022f0d7a86f982a1621a6d43624c28b0f88ddfb986caea3c71c7fec9bca62729
erasing firmware...
problem flashing firmware!
CTAP error: 0x01 - INVALID_COMMAND
Traceback (most recent call last):
  File "/home/johnny/.local/bin/solo", line 11, in <module>
    sys.exit(solo_cli())
  File "/home/johnny/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/johnny/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/johnny/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/johnny/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/johnny/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/johnny/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/johnny/.local/lib/python3.6/site-packages/solo/cli/update.py", line 212, in update
    sig = solo_client.program_file(firmware_file)
  File "/home/johnny/.local/lib/python3.6/site-packages/solo/client.py", line 328, in program_file
    self.write_flash(i, data)
  File "/home/johnny/.local/lib/python3.6/site-packages/solo/client.py", line 192, in write_flash
    self.exchange(SoloBootloader.write, addr, data)
  File "/home/johnny/.local/lib/python3.6/site-packages/solo/client.py", line 143, in exchange_hid
    data = self.send_data_hid(SoloBootloader.HIDCommandBoot, req)
  File "/home/johnny/.local/lib/python3.6/site-packages/solo/client.py", line 138, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "/home/johnny/.local/lib/python3.6/site-packages/fido2/hid.py", line 95, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND
```

szszszsz, 2019-10-24:
I think it's not in the bootloader mode. Please check that.

johnnyasantoss, 2019-10-24:
Oh :man_facepalming: 
That was the issue. Would be cool if the script could detect this and remind me to enter bootloader mode as the web version does.

0x0ece, 2019-10-24:
I changed the title, this is a good first issue. We’re going to fix it.

johnnyasantoss, 2019-10-24:
I was checking the src code and saw that there's a check (which for now is ignoring the result from the called method) for that in this section https://github.com/solokeys/solo-python/blob/4f6a09c852967db99b288c6a75c5bfc9c82b14ee/solo/cli/update.py#L92-L97 and here https://github.com/solokeys/solo-python/blob/4f6a09c852967db99b288c6a75c5bfc9c82b14ee/solo/client.py#L258-L267 this specific error is being ignored.

smrqdt, 2020-02-20:
Seems the same issue appears the other way round: when in bootloader mode there is no clear message why `solo key version` fails (led to me trying to debug solo-python while nothing was wrong)
```
% solo key version
Not using FIDO2 interface.
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/solo/client.py", line 193, in solo_version
    return self.send_data_hid(0x61, b"")
  File "/usr/lib/python3.8/site-packages/solo/client.py", line 142, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "/usr/lib/python3.8/site-packages/fido2/hid.py", line 94, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/solo/cli/key.py", line 408, in version
    res = solo.client.find(serial, udp=udp).solo_version()
  File "/usr/lib/python3.8/site-packages/solo/client.py", line 195, in solo_version
    data = self.exchange(SoloExtension.version)
  File "/usr/lib/python3.8/site-packages/solo/client.py", line 165, in exchange_u2f
    raise CtapError(ret)
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND
```

enrikb, 2020-09-23:
Both issues seem to obsolete.
Should this ticket be closed?

closed 2026-11-05 by johnnyasantoss