solokeys/solo1-cli #46: command not found: solo

MaPoLom, 2019-11-15:
Hello there,

Trying to update my solo keys I came across the problem that after installation it doesn't recognize the solo command.

Below the output from the installation script. That gave no error whatsoever.

Running:
A brand new macpro 13'' 2019

Any suggestions?

Thank You,

Maurice

> 
> mapolo@mpmac ~ % pip3 install -U solo-python
> Requirement already up-to-date: solo-python in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (0.0.19)
> Requirement already satisfied, skipping upgrade: cryptography in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (2.8)
> Requirement already satisfied, skipping upgrade: pyserial in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (3.4)
> Requirement already satisfied, skipping upgrade: fido2==0.7.3 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (0.7.3)
> Requirement already satisfied, skipping upgrade: pyusb in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (1.0.2)
> Requirement already satisfied, skipping upgrade: ecdsa in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (0.14.1)
> Requirement already satisfied, skipping upgrade: intelhex in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (2.2.1)
> Requirement already satisfied, skipping upgrade: click>=7.0 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (7.0)
> Requirement already satisfied, skipping upgrade: requests in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from solo-python) (2.22.0)
> Requirement already satisfied, skipping upgrade: six>=1.4.1 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from cryptography->solo-python) (1.13.0)
> Requirement already satisfied, skipping upgrade: cffi!=1.11.3,>=1.8 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from cryptography->solo-python) (1.13.2)
> Requirement already satisfied, skipping upgrade: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests->solo-python) (1.25.7)
> Requirement already satisfied, skipping upgrade: chardet<3.1.0,>=3.0.2 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests->solo-python) (3.0.4)
> Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests->solo-python) (2019.9.11)
> Requirement already satisfied, skipping upgrade: idna<2.9,>=2.5 in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from requests->solo-python) (2.8)
> Requirement already satisfied, skipping upgrade: pycparser in /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages (from cffi!=1.11.3,>=1.8->cryptography->solo-python) (2.19)
> mapolo@mpmac ~ % solo
> zsh: command not found: solo

nickray, 2019-11-15:
I don't have a Mac... But I assume `pip` is installing into a directory that is not in your `PATH`?

MaPoLom, 2019-11-18:
@nickray I think you are right, found bin file in some obscure folder on my system. Thank Linus for Linux or even Bill for the M-word

closed 2019-11-18 by MaPoLom