solokeys/solo1-cli #49: Doesn't work with latest fido2

veprbl, 2019-12-02:
```
    File "solo/__init__.py", line 15, in <module>
      from . import client, commands, dfu, helpers, operations
    File "solo/client.py", line 26, in <module>
      from fido2.utils import Timeout
  ImportError: cannot import name 'Timeout' from 'fido2.utils' (/nix/store/3sisrlq99lahd0dd2a057n6r8nv9iz4g-python3.7-fido2-0.8.1/lib/python3.7/site-packages/fido2/utils.py)
```
The issue with
https://github.com/solokeys/solo-python/blob/908fa6f57893f5c4b277a44c04be1e6d9e9bd250/solo/client.py#L25
seems to be that the Timeout was recently removed
https://github.com/Yubico/python-fido2/commit/4c48977173fd32f70e61f518c7624558ffe54b9c#diff-8cc874c99a54f35b68bc2633e89a5af9L125

veprbl, 2019-12-02:
This is addressed by #47

wucke13, 2020-01-03:
This should be fixed rather sooner than later IMO.  Whether #47 is going to be used or a better approach is implemented is of secondary importance, as long as this packages starts working again with the latest fido2 release.

stovocor, 2020-01-11:
I am no Python expert, but I tried to replicate the replacements made in python-fido2 when they removed the Timeout. Please see my pull request (#51).

commit b9084594 (2020-01-12): Ensure compatibility with python-fido2 0.8 (fixes #49)
commit bd1366ca (2020-01-12): Merge pull request #51 from stovocor/fido2-0.8

Ensure compatibility with python-fido2 0.8 (fixes #49)
closed 2020-01-12 by nickray