solokeys/solo1-cli #52: TypeError: make_credential() takes 2 positional arguments but 4 were given

patvdleer, 2020-01-14:
```
$ solo key verify
Please press the button on your Solo key
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/solo/cli/key.py", line 322, in verify
    cert = solo.client.find(serial, udp=udp).make_credential(pin=pin)
  File "/usr/local/lib/python3.7/dist-packages/solo/client.py", line 233, in make_credential
    rp, user, challenge, exclude_list=[], pin=pin
TypeError: make_credential() takes 2 positional arguments but 4 were given
```

patvdleer, 2020-01-14:
```
$ solo key make-credential
Touch your authenticator to generate a credential...
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/solo/cli/key.py", line 135, in make_credential
    host=host, user_id=user, serial=serial, output=True, prompt=prompt, udp=udp
  File "/usr/local/lib/python3.7/dist-packages/solo/hmac_secret.py", line 44, in make_credential
    rp, user, challenge, extensions=hmac_ext.create_dict(), pin=pin
TypeError: make_credential() takes 2 positional arguments but 4 were given
```

nickray, 2020-01-14:
I'll fix this.

nickray, 2020-01-14:
I can't reproduce. What versions of solo-python and Yubico/python-fido2 do you have installed?

s00500, 2020-01-15:
Having the same issue, solo version output: `0.0.21`

patvdleer, 2020-01-15:
Im on latest master

nickray, 2020-01-15:
I'm asking about the fido2 dependency, can you tell?

mporrato, 2020-01-15:
Same issue here with a freshly created venv:

```shell
$ python3 --version
Python 3.7.6
$ pip list
Package      Version   
------------ ----------
certifi      2019.11.28
cffi         1.13.2    
chardet      3.0.4     
Click        7.0       
cryptography 2.8       
ecdsa        0.15      
fido2        0.8.1     
idna         2.8       
intelhex     2.2.1     
pip          19.3.1    
pycparser    2.19      
pyserial     3.4       
pyusb        1.0.2     
requests     2.22.0    
setuptools   41.2.0    
six          1.13.0    
solo-python  0.0.21    
urllib3      1.25.7    
```

s00500, 2020-01-15:
```
 ~ pip3 list | grep fido2
fido2                                   0.8.1    
```

nickray, 2020-01-15:
I see now. Can you try new master please?

mporrato, 2020-01-15:
@nickray 0.0.22 works for me. Thanks!

nickray, 2020-01-15:
Thank heavens...

s00500, 2020-01-15:
Confirming, thanks for the super fast fix

mporrato, 2020-01-15:
@nickray not sure if this is related, but some commands that used to work before are now raising an AttributeError:
```
$ solo key version
Traceback (most recent call last):
  File "/home/guru/rumenta/solo-python/venv/bin/solo", line 4, in <module>
    solo_cli()
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/solo/cli/key.py", line 409, in version
    res = solo.client.find(serial, udp=udp).solo_version()
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/solo/client.py", line 193, in solo_version
    return self.send_data_hid(0x61, b"")
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/solo/client.py", line 142, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "/home/guru/rumenta/solo-python/venv/lib64/python3.7/site-packages/fido2/hid.py", line 88, in call
    while not event.is_set():
AttributeError: 'float' object has no attribute 'is_set'
```

nickray, 2020-01-15:
Oh jeez, this is utterly annoying. Pulling in the fido2 breaking changes in 0.8 were in fact breaking changes :/

nickray, 2020-01-15:
Continuing in https://github.com/solokeys/solo-python/pull/53.

mporrato, 2020-01-15:
Tried a few operations with v0.0.23: looks fine now.

nickray, 2020-01-17:
@patvdleer can you check and close if appropriate? Thanks.

closed 2020-06-18 by nickray