solokeys/solo1-cli #54: solo key verify lists Solo as Solo Tap

My1, 2020-01-17:
basically what it says in the title.

I literally just updated because the verifications were failing, but now my solo is seen as a Tap, which is weird

kinetic-algorithm0, 2020-02-12:
> 
> 
> basically what it says in the title.
> 
> I literally just updated because the verifications were failing, but now my solo is seen as a Tap, which is weird

Up-voting this issue. Verifying a newly bought regular Solo (non-NFC) updated to 3.1.0 firmware is showing as Solo Tap. 

Combing through the solo/cli/key.py file, I see the fingerprint list. Commenting out the Solo Tap fingerprint dictionary key-pair in the list causes the exception to pop, so the fingerprint value from the regular Solo is matching that entry. 

Is there a master reference for the cert fingerprints for the various SoloKey models? Just wondering if the fingerprint list needs updated. Looking at it, I would have expected a match against the regular Solo fingerprint entry. Unless I was incorrectly sold a Solo Tap 😄 (shipping packaging shows Solo USB-A though)

I don't have a Solo Tap to see what fingerprint value it pops so not sure if the verify function would return the proper match or not for that model.

philipreimer, 2020-05-23:
Same here. Bought a Solo USB-A months ago but after after updating firmware it now verifies as Solo Tap

```
$ solo key version
4.0.0 locked

$ solo key verify
Please press the button on your Solo key
Register valid
Valid Solo Tap with firmware from SoloKeys.
```