solokeys/solo1-cli #61: Fails to `solo key verify`

michaelblyons, 2020-02-24:
```
[19:04:18 ~]$ solo key version
3.1.1 unlocked
[19:04:25 ~]$ solo key verify
Please press the button on your Solo key
Unexpected run loop exit code: 3
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/fido2/_pyu2f/macos.py", line 451, in Read
    return self.read_queue.get(False)
  File "/usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/queue.py", line 167, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/solo/cli/key.py", line 322, in verify
    cert = solo.client.find(serial, udp=udp).make_credential(pin=pin)
  File "/usr/local/lib/python3.7/site-packages/solo/client.py", line 235, in make_credential
    attest, data = self.client.make_credential(options, pin=pin)
  File "/usr/local/lib/python3.7/site-packages/fido2/client.py", line 406, in make_credential
    kwargs.get("on_keepalive"),
  File "/usr/local/lib/python3.7/site-packages/fido2/client.py", line 470, in _ctap2_make_credential
    on_keepalive,
  File "/usr/local/lib/python3.7/site-packages/fido2/ctap2.py", line 705, in make_credential
    on_keepalive,
  File "/usr/local/lib/python3.7/site-packages/fido2/ctap2.py", line 642, in send_cbor
    response = self.device.call(CTAPHID.CBOR, request, event, on_keepalive)
  File "/usr/local/lib/python3.7/site-packages/fido2/hid.py", line 89, in call
    status, resp = self._dev.InternalRecv()
  File "/usr/local/lib/python3.7/site-packages/fido2/_pyu2f/hidtransport.py", line 308, in InternalRecv
    first_read = self.InternalReadFrame()
  File "/usr/local/lib/python3.7/site-packages/fido2/_pyu2f/hidtransport.py", line 300, in InternalReadFrame
    frame = self.hid_device.Read()
  File "/usr/local/lib/python3.7/site-packages/fido2/_pyu2f/macos.py", line 453, in Read
    raise OSError('Failed reading a response')
OSError: Failed reading a response
```

After this, it would not respond correctly until I pulled it out and put it back in.

nickray, 2020-03-03:
Hey @michaelblyons, I hope someone else can help, I have no access/experience with macOS :/ What does this "Cellar" do?

michaelblyons, 2020-03-04:
Cellar is a mnemonic for a beer-themed package manager ([Homebrew](https://brew.sh)) on MacOS. Macs do not come with Python 3 (although installing XCode also adds it).

It's conceivable that the `pip` I used put stuff in the wrong `site-packages`, but I think it's rather unlikely. Those point to the same place:

```
$ ls -l /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages
lrwxr-xr-x  1 me  mygroup  54 Feb 27 11:25 /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages -> ../../../../../../../../../lib/python3.7/site-packages
```

michaelblyons, 2020-06-17:
Works now. 🤷‍♂️

closed 2026-03-03 by michaelblyons