solokeys/solo1-cli #63: Unable to install

dmindfuse, 2020-03-07:
Hi. I used to get this working in a VM, but today I've been trying for 3 hours to install solo-python with no luck. I continue to run into uhid-freebsd related compilation errors. I'm on the latest Linux Mint (apt doesn't offer 3.6). I had to manually compile and install python3.6 (altinstall).

The problem I keep crashing on is:

`Installing collected packages: uhid-freebsd, fido2, intelhex, pyserial, pyusb
    Running setup.py install for uhid-freebsd ... error
    ERROR: Command errored out with exit status 1:
     command: /home/tom/git/solo-python/venv/bin/python3.6 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-g7o8t11i/uhid-freebsd/setup.py'"'"'; __file__='"'"'/tmp/pip-install-g7o8t11i/uhid-freebsd/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ygaw0jk9/install-record.txt --single-version-externally-managed --compile --install-headers /home/tom/git/solo-python/venv/include/site/python3.6/uhid-freebsd
         cwd: /tmp/pip-install-g7o8t11i/uhid-freebsd/
    Complete output (15 lines):
    running install
    running build
    running build_ext
    building 'uhid_freebsd' extension
    creating build
    creating build/temp.linux-x86_64-3.6
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -I/home/tom/git/solo-python/venv/include -I/usr/local/include/python3.6m -c uhid_freebsd.cpp -o build/temp.linux-x86_64-3.6/uhid_freebsd.o
    In file included from /usr/include/c++/5/regex:35:0,
                     from uhid_freebsd.cpp:12:
    /usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
     #error This file requires compiler and library support \
      ^
    uhid_freebsd.cpp:15:31: fatal error: dev/usb/usb_ioctl.h: No such file or directory
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/tom/git/solo-python/venv/bin/python3.6 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-g7o8t11i/uhid-freebsd/setup.py'"'"'; __file__='"'"'/tmp/pip-install-g7o8t11i/uhid-freebsd/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ygaw0jk9/install-record.txt --single-version-externally-managed --compile --install-headers /home/tom/git/solo-python/venv/include/site/python3.6/uhid-freebsd Check the logs for full command output.
Traceback (most recent call last):
  File "venv/bin/flit", line 8, in <module>
    sys.exit(main())
  File "/home/tom/git/solo-python/venv/lib/python3.6/site-packages/flit/__init__.py", line 171, in main
    pth=args.pth_file).install()
  File "/home/tom/git/solo-python/venv/lib/python3.6/site-packages/flit/install.py", line 407, in install
    self.install_directly()
  File "/home/tom/git/solo-python/venv/lib/python3.6/site-packages/flit/install.py", line 295, in install_directly
    self.install_requirements()
  File "/home/tom/git/solo-python/venv/lib/python3.6/site-packages/flit/install.py", line 249, in install_requirements
    check_call(cmd)
  File "/usr/local/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/tom/git/solo-python/venv/bin/python3.6', '-m', 'pip', 'install', '-r', '/tmp/tmpja_btl_prequirements.txt']' returned non-zero exit status 1.
Makefile:52: recipe for target 'update-venv' failed
make: *** [update-venv] Error 1
`

I setup my venv with python 3.6. The user guide on github is very unspecific about the environment. There is pip3 and pip and the commands are intertwined in the instructions and in makefile.

What is wrong up here? What's missing and how to recover? I would improve on the prerequisite instructions in the user guide if I may say so. Appreciate some guidance. Thanks.

conorpp, 2020-03-07:
Looks like `uhid-freebsd` is getting pulled in as a dependency to support HID USB communication for `fido2` library.  [From their README](https://github.com/Yubico/python-fido2#requirements): 

> Support for OpenBSD and FreeBSD is provided as-is and relies on community contributions.

It seems for some reason Python thinks it's running on BSD?  Maybe you compiled Python 3.6 with some BSD flags?

wildart, 2020-03-21:
Have the same issue
```
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial
```

```
Building wheels for collected packages: uhid-freebsd
  Running setup.py bdist_wheel for uhid-freebsd ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-f6xv6aco/uhid-freebsd/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmpyjuyc454pip-wheel- --python-tag cp35:
  /usr/lib/python3.5/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  running bdist_wheel
  running build
  running build_ext
  building 'uhid_freebsd' extension
  creating build
  creating build/temp.linux-x86_64-3.5
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c uhid_freebsd.cpp -o build/temp.linux-x86_64-3.5/uhid_freebsd.o
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  In file included from /usr/include/c++/5/regex:35:0,
                   from uhid_freebsd.cpp:12:
  /usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
   #error This file requires compiler and library support \
    ^
  uhid_freebsd.cpp:15:31: fatal error: dev/usb/usb_ioctl.h: No such file or directory
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

closed 2020-06-18 by nickray