solokeys/solo1-cli #64: Somu update not working

jakobbraun, 2020-03-08:
I've tried to update my SOMU key using this tool. But the key did reject the firmeware:
```
$ solo key update
Not using FIDO2 interface.
Wrote temporary copy of firmware-3.1.2.json to /tmp/tmpvcjvo97b.json
sha256sums coincide: 4a24fae228b5c5560d5ed8982770a72dec3cae5dcb02f6fc768b03a60afb10cf
using signature version >2.5.3
erasing firmware...
updated firmware 100%             
time: 7.45 s
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters
```
Now my key is constantly flashing red.
How can I now fix my key again? (The web updater could not fix)
Probably this tool only works for Solo keys and not for Somu? A check before update would be very helpful for preventing others to run into this issue.

conorpp, 2020-03-13:
The tool should work fine, but we've been seeing some issues with _some_ devices rejecting the latest update (3.1.2).  I'm planning to look more into this soon and fix the issue with the last update build, but in the meantime you can restore to 3.1.1 firmware.

```
wget https://github.com/solokeys/solo/releases/download/3.1.1/firmware-3.1.1.json
solo program bootloader firmware-3.1.1.json
```

jakobbraun, 2020-04-05:
Thank you for your hint. This did work. Should we close this issue in favor to a different one tracking the rejected firmware?

nickray, 2020-06-18:
Thanks for your report, please open a new issue if a specific problem persists.

closed 2020-06-18 by nickray