solokeys/solo1-cli #65: solo.exceptions.NonUniqueDeviceError

ximex, 2020-03-11:
`solo key version` etc with 2 inserted keys results in this error:

```
Traceback (most recent call last):
  File "/home/xxxx/.local/bin/solo", line 10, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/xxxx/.local/lib/python3.7/site-packages/solo/cli/key.py", line 409, in version
    res = solo.client.find(serial, udp=udp).solo_version()
  File "/home/xxxx/.local/lib/python3.7/site-packages/solo/client.py", line 46, in find
    p.find_device(dev=raw_device, solo_serial=solo_serial)
  File "/home/xxxx/.local/lib/python3.7/site-packages/solo/client.py", line 104, in find_device
    raise solo.exceptions.NonUniqueDeviceError
solo.exceptions.NonUniqueDeviceError
```

nickray, 2020-03-13:
Yeah, those stack traces sure are ugly.

You need to specify which key you want by specifying its serial number (which you can get with `solo ls`).

My1, 2020-03-27:
fun fact this issue also happens if a solo and a non-solo fido2 device is plugged into the computer at the same time, but that device isnt listed on solo ls making this a bit confusing

closed 2020-06-18 by nickray