solokeys/solo1-cli #69: graceful erroring

My1, 2020-03-25:
currently many errors give out a stacktrace, which for example looks like this:

```
my1@my1-qb:~$ solo key make-credential
Touch your authenticator to generate a credential...
Traceback (most recent call last):
  File "/home/my1/.local/bin/solo", line 11, in <module>
    sys.exit(solo_cli())
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/my1/.local/lib/python3.6/site-packages/solo/cli/key.py", line 135, in make_credential
    host=host, user_id=user, serial=serial, output=True, prompt=prompt, udp=udp
  File "/home/my1/.local/lib/python3.6/site-packages/solo/hmac_secret.py", line 44, in make_credential
    rp, user, challenge, extensions=hmac_ext.create_dict(), pin=pin
  File "/home/my1/.local/lib/python3.6/site-packages/fido2/client.py", line 339, in make_credential
    on_keepalive,
  File "/home/my1/.local/lib/python3.6/site-packages/fido2/client.py", line 369, in _ctap2_make_credential
    raise ValueError("PIN required!")
ValueError: PIN required!
```

or this
```
my1@my1-qb:~$ solo key make-credential
Traceback (most recent call last):
  File "/home/my1/.local/bin/solo", line 11, in <module>
    sys.exit(solo_cli())
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/my1/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke                                                                                                                                         
    return callback(*args, **kwargs)                                                                                                                                                                                             
  File "/home/my1/.local/lib/python3.6/site-packages/solo/cli/key.py", line 135, in make_credential                                                                                                                              
    host=host, user_id=user, serial=serial, output=True, prompt=prompt, udp=udp                                                                                                                                                  
  File "/home/my1/.local/lib/python3.6/site-packages/solo/hmac_secret.py", line 30, in make_credential                                                                                                                           
    client = solo.client.find(solo_serial=serial, udp=udp).client                                                                                                                                                                
  File "/home/my1/.local/lib/python3.6/site-packages/solo/client.py", line 46, in find                                                                                                                                           
    p.find_device(dev=raw_device, solo_serial=solo_serial)                                                                                                                                                                       
  File "/home/my1/.local/lib/python3.6/site-packages/solo/client.py", line 104, in find_device                                                                                                                                   
    raise solo.exceptions.NonUniqueDeviceError                                                                                                                                                                                   
solo.exceptions.NonUniqueDeviceError
```

instead of using those stack traces for "simple" errors like that a PIN is needed, it would be nice to error out more gracefully, which could be done 2 ways:

1) remove the big trace and add a pretty error message like "you have multiple Solos connected, please only connect only one solom or specify which to use by `-s serial` or whatever)
2) auto-guide the user through, like in the case of multiple solos offer to tap one or enter the serial number directly while listing the ones currently connected

nickray, 2020-03-26:
Yeah, with `click` we have a simple way to expose functionality, but the stack traces aren't great. Basically, every (sub)command needs to catch the possible errors, and output a user-friendly error instead. It's probably best to do this by explicitly catching *expected* errors only.

My1, 2020-03-26:
sadly I am not a python or desktop programmer in general. so I can mostly point out things that dont work or give Ideas. but yeah throwing out the ugly stuff for the most common errors probebly helps

My1, 2020-03-29:
PIN stuff should be handled by #76 unless I missed something

closed 2020-06-18 by nickray