solokeys/solo1-cli #77: show release notes on update

riedel, 2020-03-29:
I just updated the firmware reading that it might erase all RKs (which is fine to me). However, I wondered if that is an expected behaviour if another user is calling 

> solo key update

My suggestion would be to actually add notices to the update json and display them on the commandline...

nickray, 2020-04-01:
Yes, we should prevent major version updates with non-empty keys.

riedel, 2020-04-03:
So the workflow would be reset first then update major version? 
Will major versions always break resident keys in the future?

 It would be quite convienient if they would not to "dirty flash". If that is not sure, I would issue just capital letter warnings. IMHO it is not good style to patronize users too heavily.

What shouldn't happen is that everything breaks without a word of warning

nickray, 2020-04-26:
There's no reason for a major version update if it's backwards-compatible, so I'd say "yes".

closed 2020-06-18 by nickray