solokeys/solo1-cli #78: HidOverUDP

h8h, 2020-04-21:
I wonder how to use [HidOverUDP](https://github.com/solokeys/solo-python/blob/master/solo/fido2/__init__.py#L17) to detect software solos.

I tried the "software" way and my solo sw is listening on udp/8111.
```
~$ solo version
0.0.24
```

```
solo$ ./main
[DEBUG] Using UDP backing
state file exists
[ERR] Current firmware version address: 0x556e63ef61f0
[ERR] Current firmware version: 4.0.0.0 (04.00.00.00)
[STOR] Auth state is initialized
[STOR] pin not set.
```
```
~$ ss -uapn
State  Recv-Q Send-Q          Local Address:Port     Peer Address:Port Process                         
UNCONN 0      0                     0.0.0.0:8111          0.0.0.0:*     users:(("main",pid=4997,fd=3)
```

```
>>> from solo import fido2
>>> fido2.force_udp_backend()
```
```   
>>> import solo
>>> solo.client.find()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/h8h/.local/lib/python3.8/site-packages/solo/client.py", line 52, in find
    raise solo.exceptions.NoSoloFoundError("no Solo found")
solo.exceptions.NoSoloFoundError: no Solo found
```

The following works:
```
>>> a = fido2.HidOverUDP("localhost:8111")
>>> a.Write(0xff)
```
I received:
```
[ERR] ctaphid.c:500: ignoring random cont packet from 0000
```
But this is not very convenient :)

conorpp, 2020-04-21:
I'm not sure I understand.  why isn't it convenient?  What is it not convenient for?

h8h, 2020-04-22:
Excuse me for not making myself clear.

I want to use python-solo with all of its commands like

```
solo ls  # lists all Solo keys connected to your machine
solo version  # outputs version of installed `solo` library and tool
```

closed 2020-06-18 by nickray