solokeys/solo1-cli #82: Is there a way to use solo key challenge-response without having to press the solo key?

Iolaum, 2020-05-16:
When using the `solo key challenge-response` command you have to press the solo key before you get a response. Is there a way to do get the response without having to press the solo key?

I gave it a try from python as well:

```
>>>import solo.hmac_secret as shc
>>> shc.simple_secret(
...     credential_id='appropariate_credential_id',
...     secret_input='myinput',
...     host='myhost.org'
... )
Touch your authenticator to generate a reponse...
...
```
but I still had to press the solo key and couldn't find any option to circumvent it. Is there such an option?

nickray, 2020-05-17:
You need to run in "silent authenticator" mode for this. On a hacker key, you can change a flag. This is pretty unsafe though as it deactivates "user presence" tests on all keys, ideally the flag could be set by perhaps `host`? I think though that for typical use cases for this feature you'd want UP test too..

Iolaum, 2020-05-20:
Being able to authenticate silently on, say silent.mydomain.tld, sound cool. I don't think I  know enough to try to build something like that and try that. Thanks for the suggestion though.

P.S. Closing the issue since it received feedback but I don't expect much follow up on it atm.

closed 2020-05-20 by Iolaum