solokeys/solo1-cli #89: "solo key verify" causes NonUniqueDeviceError exception when Yubikey also installed

JohnVillalovos, 2020-07-09:
I have a Solo key and Yubikey 5 plugged in at the same time:

```
$ solo key verify
Please press the button on your Solo key
Traceback (most recent call last):
  File "/home/solo-user/.local/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/home/solo-user/.local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/solo-user/.local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/solo-user/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/solo-user/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/solo-user/.local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/solo-user/.local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/solo-user/.local/lib/python3.8/site-packages/solo/cli/key.py", line 323, in verify
    cert = solo.client.find(serial, udp=udp).make_credential(pin=pin)
  File "/home/solo-user/.local/lib/python3.8/site-packages/solo/client.py", line 46, in find
    p.find_device(dev=raw_device, solo_serial=solo_serial)
  File "/home/solo-user/.local/lib/python3.8/site-packages/solo/client.py", line 104, in find_device
    raise solo.exceptions.NonUniqueDeviceError
solo.exceptions.NonUniqueDeviceError
```

Vash63, 2022-02-16:
Just ran into this myself on the latest solo-python, didn't realize it was the Yubi causing the issue until I found this issue report.

nickray, 2022-03-06:
The stack traces aren't pretty... Maybe you can `solo ls` first and then select which device to verify via flag.