solokeys/solo1-cli #90: 'solo key update' complains about multiple solo keys

JohnVillalovos, 2020-07-10:
I have one solo key installed and one Yubikey 5 key installed.

This happens:

```
$ solo key update

Multiple Solo keys are plugged in! Please:
  * unplug all but one key
```

197788, 2021-12-21:
Same here. 

> 

> I have one solo key installed and one Yubikey 5 key installed.
> 
> This happens:
> 
> ```
> $ solo key update
> 
> Multiple Solo keys are plugged in! Please:
>   * unplug all but one key
> ```
> 
> When I removed the Yubikey the error went away.

auwsom, 2022-04-16:
Me too. Unplugged the Yubikey and update worked.

I got the same output from `solo1 ls` though in both states: 

```
solo1 ls
:: Solos
########: SoloKeys Solo 4.0.0
```