solokeys/solo1-cli #94: solokey for LUKS encryption

saravanan30erd, 2020-08-18:
I am trying to use its challenge response method for unlocking a LUKS-encrypted drive (encrypt complete / partition except /boot partition).

I tried to create custom hook(in Arch Linux) for solokey, challenge-response will be performed during booting sequence.
The solokey tool is a python package, I tried to include the python & solo in custom hook on initrd image(using mkinitcpio) but facing issues related to python.

This is already done for Yubikey - https://github.com/agherzan/yubikey-full-disk-encryption
yubikey hook - https://github.com/agherzan/yubikey-full-disk-encryption/blob/master/src/install/ykfde

Since yubikey (ykchalresp) tool is a simple binary, its straightforward to include ykchalresp in initrd image. But solokey is only available as python package, facing difficulty including python&solo in initrd image for Arch Linux.
https://github.com/saravanan30erd/solokey-full-disk-encryption/blob/master/skfde-install

Is it any other tool(simple binary same as ykchalresp)  for solokey?

nickray, 2020-08-23:
I have no LUKS experience, but I know that https://github.com/shimunn/fido2luks exists.

saravanan30erd, 2020-08-24:
@nickray thanks, I used this tool to create hook for initrd in Arch linux.

https://github.com/saravanan30erd/solokey-full-disk-encryption/blob/master/hooks/skfde

Looks like solokey is not detected in initrd stage, getting `No such device or address (os error 6)`

saravanan30erd, 2020-08-27:
issue related to the tool, not about the device. closing the issue

closed 2020-08-27 by saravanan30erd

coelner, 2021-01-30:
FYI: http://0pointer.net/blog/unlocking-luks2-volumes-with-tpm2-fido2-pkcs11-security-hardware-on-systemd-248.html