solokeys/solo1-cli #96: Can't use feedkernel; doesn't allow running as root

jscott0, 2020-09-03:
Hi,

I'm trying out the solo-python package for Debian, version 0.0.26, and I can't use feedkernel.
```
$ solo key rng feedkernel
Entropy before: 0x3866
Traceback (most recent call last):
File "/usr/bin/solo", line 4, in <module>
solo_cli()
File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
return self.main(*args, **kwargs)
File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
rv = self.invoke(ctx)
File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
return callback(*args, **kwargs)
File "/usr/lib/python3/dist-packages/solo/cli/key.py", line 117, in feedkernel
fcntl.ioctl(fh, RNDADDENTROPY, t)
PermissionError: [Errno 1] Operation not permitted
```

It refuses to let me use it as root:
```
$ sudo solo key rng feedkernel
THIS COMMAND SHOULD NOT BE RUN AS ROOT!

Please install udev rules and run `solo` as regular user (without sudo).
For more information, see: https://docs.solokeys.io/solo/udev
Entropy before: 0x3911
Entropy after:  0x3915
```

I do have the udev rules though. The merge request #18 that introduced the feedkernel option says that it should just warn in recent versions, but 0.0.26 is the latest.

enrikb, 2020-09-23:
You can set the environment variable `ALLOW_ROOT` to suppress the warning if you know what you're doing.
```
$ sudo ALLOW_ROOT=1 solo key rng feedkernel
```

197788, 2021-12-21:
Here is what I get after trying to run solo key rng feedkernel:

Traceback (most recent call last):
File "/usr/bin/solo", line 8, in <module>
sys.exit(solo_cli())
File "/usr/lib/python3.10/site-packages/click/core.py", line 1128, in __call__
return self.main(*args, **kwargs)
File "/usr/lib/python3.10/site-packages/click/core.py", line 1053, in main
rv = self.invoke(ctx)
File "/usr/lib/python3.10/site-packages/click/core.py", line 1659, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.10/site-packages/click/core.py", line 1659, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.10/site-packages/click/core.py", line 1659, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.10/site-packages/click/core.py", line 1395, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/usr/lib/python3.10/site-packages/click/core.py", line 754, in invoke
return __callback(*args, **kwargs)
File "/usr/lib/python3.10/site-packages/solo/cli/key.py", line 83, in feedkernel
p = solo.client.find(serial)
File "/usr/lib/python3.10/site-packages/solo/client.py", line 33, in find
p.find_device(dev=raw_device, solo_serial=solo_serial)
File "/usr/lib/python3.10/site-packages/solo/devices/solo_v1.py", line 63, in find_device
raise solo.exceptions.NonUniqueDeviceError
solo.exceptions.NonUniqueDeviceError