solokeys/solo1-cli #110: Installing fails on ubuntu-20.04 when fido2 0.9.0 is used

uli-heller, 2021-01-20:
I tried to install on an ubuntu-20.04 box. I'm doing this:

```
uli@ulicsl:~/git/cloned$ pip3 install solo-python
Defaulting to user installation because normal site-packages is not writeable
Collecting solo-python
  Downloading solo_python-0.0.26-py3-none-any.whl (43 kB)
     |████████████████████████████████| 43 kB 1.2 MB/s 
Requirement already satisfied: click>=7.0 in /usr/lib/python3/dist-packages (from solo-python) (7.0)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from solo-python) (2.22.0)
Collecting fido2>=0.8
  Downloading fido2-0.9.0.tar.gz (206 kB)
     |████████████████████████████████| 206 kB 2.7 MB/s 
Collecting pyserial
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     |████████████████████████████████| 90 kB 5.2 MB/s 
Collecting pyusb
  Downloading pyusb-1.1.1-py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 5.6 MB/s 
Collecting ecdsa
  Downloading ecdsa-0.16.1-py2.py3-none-any.whl (104 kB)
     |████████████████████████████████| 104 kB 8.8 MB/s 
Requirement already satisfied: cryptography in /usr/lib/python3/dist-packages (from solo-python) (2.8)
Collecting intelhex
  Downloading intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
     |████████████████████████████████| 50 kB 6.8 MB/s 
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from fido2>=0.8->solo-python) (1.14.0)
Building wheels for collected packages: fido2
  Building wheel for fido2 (setup.py) ... done
  Created wheel for fido2: filename=fido2-0.9.0-py2.py3-none-any.whl size=178439 sha256=cd89cfc7ae5485140714b8e1e033f710fc5e7d8adc9e375591ab30f67525b57f
  Stored in directory: /home/uli/.cache/pip/wheels/09/9c/f7/81b6838898d93db1c11373e023bcebc1261b66455ccba0c4c8
Successfully built fido2
Installing collected packages: fido2, pyserial, pyusb, ecdsa, intelhex, solo-python
Successfully installed ecdsa-0.16.1 fido2-0.9.0 intelhex-2.3.0 pyserial-3.5 pyusb-1.1.1 solo-python-0.0.26
```

I'm getting this error:

```
uli@ulicsl:~/git/cloned$ solo --help
Traceback (most recent call last):
  File "/home/uli/.local/bin/solo", line 5, in <module>
    from solo.cli import solo_cli
  File "/home/uli/.local/lib/python3.8/site-packages/solo/cli/__init__.py", line 17, in <module>
    from solo.cli.key import key
  File "/home/uli/.local/lib/python3.8/site-packages/solo/cli/key.py", line 24, in <module>
    import solo.fido2
  File "/home/uli/.local/lib/python3.8/site-packages/solo/fido2/__init__.py", line 3, in <module>
    import fido2._pyu2f
ModuleNotFoundError: No module named 'fido2._pyu2f'
```

Downgrading to 0.8.1 fixes the issue:

```
uli@ulicsl:~/git/cloned$ pip3 install "fido2==0.8.1"
Defaulting to user installation because normal site-packages is not writeable
Collecting fido2==0.8.1
  Downloading fido2-0.8.1.tar.gz (201 kB)
     |████████████████████████████████| 201 kB 2.9 MB/s 
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from fido2==0.8.1) (1.14.0)
Requirement already satisfied: cryptography>=1.5 in /usr/lib/python3/dist-packages (from fido2==0.8.1) (2.8)
Building wheels for collected packages: fido2
  Building wheel for fido2 (setup.py) ... done
  Created wheel for fido2: filename=fido2-0.8.1-py2.py3-none-any.whl size=160370 sha256=7709875821d6129fcbe2a668c57741191d5e626fd42fe7e3f40cd5be1c5e191c
  Stored in directory: /home/uli/.cache/pip/wheels/cd/28/c1/f0805b67ded1cca2509030ad301b90ce9f39b1be26422b4079
Successfully built fido2
Installing collected packages: fido2
  Attempting uninstall: fido2
    Found existing installation: fido2 0.9.0
    Uninstalling fido2-0.9.0:
      Successfully uninstalled fido2-0.9.0
Successfully installed fido2-0.8.1

uli@ulicsl:~/git/cloned$ solo --help
Usage: solo [OPTIONS] COMMAND [ARGS]...

Options:
  --help  Show this message and exit.

Commands:
  genkey    Generates key pair that can be used for Solo signed firmware...
  key       Interact with Solo keys, see subcommands.
  ls        List Solos (in firmware or bootloader mode) and potential Solos...
  mergehex  Merges hex files, and patches in the attestation key.
  monitor   Reads Solo Hacker serial output from USB serial port...
  program   Program a key.
  sign      Signs a firmware hex file, outputs a .json file that can be
            used...
  version   Version of python-solo library and tool.
```

uli-heller, 2021-01-20:
I got this: https://github.com/Yubico/python-fido2/issues/109

```
0.9 is not backwards compatible with 0.8. You'll have to open an issue with the python-solo project about updating their code to support 0.9, or to pin the dependency version to 0.8.1.
```

closed 2021-01-20 by nickray

innir, 2021-03-19:
Any plans for porting to python-fido2 0.9.x?
I packaged solo-python for Debian (https://tracker.debian.org/pkg/solo-python) but as python-fido2 was updated to 0.9.1 it's broken at the moment, see https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=985462 :-/

nickray, 2021-03-19:
There are no immediate plans on SoloKeys side due to limited dev resources, however @pilou- has announced work being done (https://github.com/solokeys/solo-python/pull/118#issuecomment-797655040).

Is the Debian freeze in effect? I was not aware. If not, is there a realistic timeline? I understand python-fido2 is trying to move to a 1.0 next, at which point we'd revisit at the latest.

Independently, I'd be interested in a chat with you, as the new key will have a Rust CLI that would be great to package - and I don't understand the Debian process for that.

innir, 2021-04-19:
@nickray Sorry for coming back to you so late. Concerning Rust packaging in Debian, I think the Rust Packaging Team is the best starting point: https://wiki.debian.org/Teams/RustPackaging.

Debian is frozen (see https://release.debian.org/bullseye/freeze_policy.html) but bug fixes are still possible - so we would need a patch that just makes the needed changes to work with python-fido2 0.9.x but I fear @pilou- lost interest / doesn't have time to complete his patch (see https://github.com/solokeys/solo-python/pull/118#issuecomment-808721750)

innir, 2021-04-25:
Just a short update: If there is no fix within the next 14 days solo-python will not be part of Debian Bullseye unfortunately :-/ (@nickray @pilou- )

nickray, 2021-04-25:
@innir Thanks for the heads-up, I'll try and give it a go.

nickray, 2021-04-25:
@innir this update on our side would be much simplified if https://github.com/Yubico/python-fido2/pull/113 got a release in. Is there any chance of that happening? Presumably not since even if upstream releases, it wouldn't be a bugfix?

innir, 2021-04-25:
@nickray I don't think it is possible to get a new version of python-fido2 into Debian Bullseye - as you said, it's not a bugfix but a new release :-/

nickray, 2021-04-25:
We're looking into it. Meanwhile, as ultima ratio... is pipx going to be in Debian Bullseye?

innir, 2021-04-25:
Thanks for your effort! Unfortunately pipx will not be part of Bullseye: https://tracker.debian.org/pkg/python-pipx

nickray, 2021-04-25:
@innir Can you check if release 0.0.28 integrates with Debian as expected?

conorpp, 2021-04-25:
I made a new release, 0.0.28, that uses 0.9.1.

https://github.com/solokeys/solo-python/pull/121

innir, 2021-04-26:
Wow, thanks, that was quick! I tested 0.0.28 (from PyPi) and it works on Debian testing (the to be Bullseye) :-) Great work!
To have a chance to get it into Bullseye we now need a patch that just implements the changes for fido2 0.9.1 compatibility on top of 0.0.27.
I already tried a little but was not successful :-/ Seems like a lot from client.py got moved to devices/* which makes the patch look huge and seems a little unrelated to the fido2 changes ... The release team is very picky at things like that - they review the whole patches and only want stuff in there that's needed to fix the bug (crashing with fido2 0.9.1).
I can understand that's kind of frustrating for you as it's extra work with no clear benefit :-/

stefanor, 2021-05-21:
@innir: Given it's a leaf package, and the patch isn't easily backportable, you can try asking the release team for a pre-emptive unblock of the new upstream release. Worst case scenario, it doesn't make it into bullseye.

PR #111 (2021-01-20): Don't load fido2-0.9.0
Maybe fixes #110 ?

I don't know python too well, so this might be complete trash. Somehow, I was not able to specify the dependency like this:
`>= 0.8, <0.9` or maybe `=^0.8.1`