solokeys/solo1-cli #117: no interaction with key possible

Kabbone, 2021-03-05:
I wanted to do firmware updates to some SOMUs when I realized solo-python seems to not work anymore.
I'm very curious what caused this, because it is running in a venv, so I suspect something else outside the python packages, but I could't figure it out yet.

`$ solo version

0.0.27`

`$ solo ls

Traceback (most recent call last):
  File "/home/kabbone/IT/Solokey/solo-python/bin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/solo/cli/__init__.py", line 136, in ls
    solos = solo.client.find_all()
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/solo/client.py", line 57, in find_all
    hid_devices = list(CtapHidDevice.list_devices())
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/fido2/hid.py", line 135, in list_devices
    for d in hidtransport.hid.Enumerate():
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/fido2/_pyu2f/linux.py", line 194, in Enumerate
    ParseUevent(uevent_file.read(), desc)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/solo/cli/_patches.py", line 113, in newParseUevent
    oldnewParseUevent(uevent, desc)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/fido2/_pyu2f/linux.py", line 164, in ParseUevent
    k, v = line.split(b'=')
ValueError: not enough values to unpack (expected 2, got 1)
`

`$ pip list

Package      Version
------------ ---------
certifi      2020.12.5
cffi         1.14.5
chardet      4.0.0
click        7.1.2
cryptography 3.4.6
ecdsa        0.16.1
fido2        0.8.1
idna         2.10
intelhex     2.3.0
pip          20.2.3
pycparser    2.20
pyserial     3.5
pyusb        1.1.1
requests     2.25.1
setuptools   49.2.1
six          1.15.0
solo-python  0.0.27
urllib3      1.26.3
`

You will already run into this with the minimal example
`Python 3.9.2 (default, Feb 20 2021, 18:40:11)
[GCC 10.2.0] on linux
import solo
solo.client.find()

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/solo/client.py", line 47, in find
    p.find_device(dev=raw_device, solo_serial=solo_serial)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/solo/client.py", line 99, in find_device
    devices = list(CtapHidDevice.list_devices())
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/fido2/hid.py", line 135, in list_devices
    for d in hidtransport.hid.Enumerate():
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/fido2/_pyu2f/linux.py", line 194, in Enumerate
    ParseUevent(uevent_file.read(), desc)
  File "/home/kabbone/IT/Solokey/solo-python/lib/python3.9/site-packages/fido2/_pyu2f/linux.py", line 164, in ParseUevent
    k, v = line.split(b'=')
ValueError: not enough values to unpack (expected 2, got 1)
`

enrikb, 2021-03-06:
AFAICS, this means that you have a HID device in your system having in its `device/uevent` sys-file a line which misses an '=' character. Not sure how this can happen. You could check this first, like:

```
cat /sys/class/hidraw/hidraw*/device/uevent
```

If you see a device with such a line and you can temporarily remove it, try this. If it happens to be the Solo key itself, the key might have an issue.

Next, you could try to change `fido2/_pyu2f/linux.py` to ignore the issue and continue, e.g. catch and ignore the exception.

Kabbone, 2021-03-06:
I already assumed that something is messed up at the split, but I thought perhaps I don't get the correct informations beforehand.

Long story short, your assumption was correct, it's my keyboard (AnnePro2). It represents its name without any "="

Thank you! Seems to work as expected again. Should I close it should it get wrapped with a try except for this rare cases?

enrikb, 2021-03-06:
As this seems to be an issue related to the python fido2 package (0.8.1), which has been replaced upstream by a newer, incompatible version which is not yet supported by solo-python IIRC, you might be best off with a local fix for your issue until solo-python might get an update.
Of course, you can also report this issue to the fido2-python project.

enrikb, 2021-03-06:
Just found that solo-python already 'live patches' the method in question. I added a very rough workaround that might fix your problem on this branch:
https://github.com/enrikb/solo-python/tree/robustness/ignore_ValueError_from_fido2_ParseUevent
Maybe you want to try it.

Kabbone, 2021-03-06:
Thanks for the fast response. I saw the patch file before, just didn't try to do myself.
Your proposed change is not enough, because after the return it's still missing, so the second split also needs the exception.
I sent you a pull request on your proposed change. With theses two changes it's working.

enrikb, 2021-03-06:
Thanks! I had it in my head but forgot to type ;-) (And didn't have a test case...)

nickray, 2021-03-12:
I'm not quite following - would any of this be fixed by updating `solo-python` to `python-fido2` v0.9.1?

enrikb, 2021-03-12:
I think the issue can be fixed by an update to v0.9.1, because it accesses the HID interfaces on Linux quite differently - at least this is my understanding after quickly checking the differences. However, updating might be quite a bit of work due to these changes.

PR #118 (2021-03-12): Ignore ValueError from fido2 ParseUevent.
As reported in #117, there can be lines in Linux hidraw sysfs files for certain devices (maybe broken? or is it a kernel issue?) missing an '=' sign.

The fido2 ParseUevent() method fails if it hits a line missing the '=' in the uevent sysfs file and `solo` aborts due to an unhandled exception.

As solo-python already live patches the method, add a try-except to the call of the original method and ignore the ValueError thrown in such a situation. The same is applied to the split() call in this method.

Closes #117.