solokeys/solo1-cli #128: solo key verify   AttributeError: 'tuple' object has no attribute 'attestation_object'

hallbw, 2021-04-29:
> solo key verify
Please press the button on your Solo key
Traceback (most recent call last):
  File "/sbin/solo", line 8, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/solo/cli/key.py", line 363, in verify
    cert = key.make_credential(pin=pin)
  File "/usr/lib/python3.9/site-packages/solo/devices/base.py", line 101, in make_credential
    attest = result.attestation_object
AttributeError: 'tuple' object has no attribute 'attestation_object'

Arch Linux
solo-python 0.0.30-1
> solo version
0.0.30
> solo key version
4.1.2 locked

conorpp, 2021-04-29:
What is your version of fido2?  `pip3 show fido2`.  Should be 0.9.1 or later.

hallbw, 2021-04-29:
That was the problem, thank you! I uninstalled my user-local installed fido2, and now the system-installed one of the proper version is seen.

> pip3 show fido2
Name: fido2
Version: 0.8.1
Summary: Python based FIDO 2.0 library
Home-page: https://github.com/Yubico/python-fido2
Author: Dain Nilsson
Author-email: dain@yubico.com
License: UNKNOWN
Requires: six, cryptography
Required-by: solo-python

pacman -Q | grep python-fido2
python-fido2 0.9.1-1

pip3 uninstall fido2

solo key verify
Please press the button on your Solo key
Register valid
Valid Solo (<=3.0.0) firmware from SoloKeys.

closed 2021-04-29 by hallbw