solokeys/solo1-cli #130: fido2.ctap.CtapError: CTAP error: 0x35 - PIN_NOT_SET

AWHubGit, 2021-05-02:
Valid Solo - FIDO2 (USB only) USB-A with firmware 4.1.2 locked from SoloKeys on ArchLinux solo-pythob version 0.0.30.
"solo key reset" was done. No PIN.
"solo key credential info" and "solo key credential ls" asking for PIN and failing with "fido2.ctap.CtapError: CTAP error: 0x35 - PIN_NOT_SET"

nickray, 2021-05-02:
Yeah credential management is only allowed when PIN is set.

coelner, 2021-12-27:
partly FYI: https://github.com/solokeys/solo/issues/570

If I'm understanding this correctly ( https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#authenticatorGetInfo) the ClientPin option shows us whether a pin is set or not.

> Client PIN:
> If present and set to true, it indicates that the device is capable of accepting a PIN from the client and PIN has been set.
> If present and set to false, it indicates that the device is capable of accepting a PIN from the client and PIN has not been set yet.
> If absent, it indicates that the device is not capable of accepting a PIN from the client.
> Client PIN is one of the ways to do user verification.

Here we check regardless of an available pin [solo key credential, solo key challenge-response]:
https://github.com/solokeys/solo-python/blob/f087629a8f17cf9de79832bda242057aae4a2475/solo/cli/key.py#L147
https://github.com/solokeys/solo-python/blob/f087629a8f17cf9de79832bda242057aae4a2475/solo/cli/key.py#L197
https://github.com/solokeys/solo-python/blob/f087629a8f17cf9de79832bda242057aae4a2475/solo/cli/key.py#L530
https://github.com/solokeys/solo-python/blob/f087629a8f17cf9de79832bda242057aae4a2475/solo/cli/key.py#L550
https://github.com/solokeys/solo-python/blob/f087629a8f17cf9de79832bda242057aae4a2475/solo/cli/key.py#L586

Here we do not check at all [solo key change-pin]:
https://github.com/solokeys/solo-python/blob/f087629a8f17cf9de79832bda242057aae4a2475/solo/cli/key.py#L310

But here it is checked [solo key verify]:
https://github.com/solokeys/solo-python/blob/f087629a8f17cf9de79832bda242057aae4a2475/solo/cli/key.py#L352-L359

Maybe it is useful to hide those entries which are not usable.