solokeys/solo1-cli #134: solo key make-credential

arvicrivera, 2021-07-12:
I opened powershell (Administrator mode), ran `solo key make-credential`, solo ask for my PIN, I enter my PIN, and this is what happens:

`PS C:\Users> solo key make-credential
PIN (leave empty for no PIN):
Touch your authenticator to generate a credential...
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Scripts\solo.exe\__main__.py", line 7, in <module>
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\click\core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\click\core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\click\core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\click\core.py", line 1668, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\click\core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\solo\cli\key.py", line 151, in make_credential
    solo.hmac_secret.make_credential(
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\solo\hmac_secret.py", line 42, in make_credential
    attestation_object = client.make_credential(
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\fido2\client.py", line 455, in make_credential
    att_obj, extension_outputs = self._do_make_credential(
  File "C:\Users\arivera\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\fido2\client.py", line 503, in _ctap2_make_credential
    raise ClientError.ERR.BAD_REQUEST("PIN required but not provided")
fido2.client.ClientError: (<ERR.BAD_REQUEST: 2>, 'PIN required but not provided')`

I know it says "PIN required but not provided", but I tried like 10 times already and always the same error.

OS: Windows 10 Enterprise
OS Version: 2004
SoloKey USB-A
Firmware version: 4.1.2 locked

Thank you in advance!

enrikb, 2021-08-04:
This should have been fixed by #133. Unfortunately not yet released, so you have to install from here.

b90g, 2021-10-17:
Why isnt it released yet?... its been 4 month and not everybody looks up github issues when they encounter a problem with their device... please fix this (release it..)