solokeys/solo1-cli #139: Update documentation for SoloKey v2

ajaust, 2021-11-10:
I was wondering whether the `solo-python` package should work for the SoloKey v2. I just got some of my keys and was immediately playing around with them. However, to me it is not clear whether `solo-python` is supposed to be used with the SoloKey v2. A quick check at least showed that it seems to be compatible. I was not able to test the firmware upgrade process though since I was not able to boot my sticks in bootloader mode yet.

I would suggest to add a statement in the documentation/README whether `solo-python` supports or aims to support the SoloKey v2 in the near future.

nickray, 2022-03-08:
I've changed the repo name and added a comment.

closed 2022-03-08 by nickray

ajaust, 2022-03-08:
Great, thanks!