solokeys/solo1-cli #151: running after a pip3 install and make init both fail after recent update to fido2 library

kms15, 2022-06-09:
The fido2 library has renamed the CTAP1 class to Ctap1 and CTAP2 class to Ctap2, deprecating the old names, and they appear to have removed the old names in the newest version of the fido2 library (currently 1.0.0). 

This causes errors when using "make init" or trying to run the solo1 cli after installing it with pip3, as it will get the latest version of the fido2 library and crash with the following error:

    ImportError: cannot import name 'CTAP1' from 'fido2.ctap1' (/home/kms15/git/solo1-cli/venv/lib/python3.10/site-packages/fido2/ctap1.py)

Pinning the fido2 package to version 0.9.2 fixes the crash, but probably the solo1-cli code should be updated to match the most recent (and hopefully stable) API of the fido2 library.

gunterkoenigsmann, 2022-06-10:
For me the linked pull request resolves this issue and #153.

hasufell, 2022-12-01:
Half a year later, this project is still broken on pip.

sseide, 2022-12-01:
applying the patch from @wuxxin fixed the issue. Can Solokeys please merge it and release a new version? should be 10 minutes work...
@0x0ece @conorpp @hoos97 @nickray

johnnyasantoss, 2022-12-13:
Just downloaded in a new computer and got this error again (I remember reading this thread before)

jhuylebroeck, 2023-02-15:
I'm not a coder myself, and I had a hard time resolving this issue. Thanks for the workaround, but I'd love a full update and solution.

AWHubGit, 2023-02-20:
> I'm not a coder myself, and I had a hard time resolving this issue. Thanks for the workaround, but I'd love a full update and solution.

Me too. Manually pinning fido2 using pip3 install fido2==0.9.3 makes cli working again, but I would like the latest version of the fido2 library.

Is it a secuirity flaw  still to have CTAP1 class?

shetyeshail, 2023-02-26:
I'm running into this issue as well. Pinning the pip fido2 library version to 0.9.3 as suggested by johnnyasantoss worked for me, but would appreciate an official patch to the solo1-cli tool. I believe the pull request from wuxxin would do the trick.

GoetzGoerisch, 2023-04-21:
Ping @0x0ece @conorpp @hoos97 @nickray 
Please update the current solo1-cli version. Thank you.

tomash, 2023-11-24:
I'm also asking to update the current solo1-cli version, manual search-and-replace in installed pip package (or fixing fido2 to an old version) is a poor experience.

Jookia, 2024-10-11:
Just for anyone interested, this lets you install solo1 using pipx:

```
pipx install --preinstall "fido2==0.9.3" solo1
```

johnnyasantoss, 2026-11-05:
using uvx (if anyone is using this in 2026)
`uvx -w "fido2~=0.9" solo1`

btw, RIP solo

tomash, 2026-11-07:
@johnnyasantoss why RIP? they seem to be progressing? https://github.com/solokeys/solo2/discussions/205

Jookia, 2026-11-08:
Oh wow, I didn't see that. That's sad news.