solokeys/solo1-cli #157: cannot set pin using solo1 due to "'Fido2Client' object has no attribute 'client_pin'"

vladak, 2022-06-23:
### Discussed in https://github.com/solokeys/solo1-cli/discussions/156

<div type='discussions-op-text'>

<sup>Originally posted by **vladak** June 23, 2022</sup>
solo1-cli version 0.1.1 (with changes matching PR #153), and brand new Solokey with updated firmware:
```
$ solo1 ls
:: Solos
None: SoloKeys Solo 4.1.5
```
and it is not possible to change the default PIN:
```
$ solo1 key change-pin
Please enter old pin: 
Please enter new pin: 
Please confirm new pin: 
'Fido2Client' object has no attribute 'client_pin'

$ solo1 key set-pin
Please enter new pin: 
Please confirm new pin: 
'Fido2Client' object has no attribute 'client_pin'
```
I installed the `solo1` via the usual means (Python venv + pip install).

The `fido2` package is at version 1.0.0.</div>

vladak, 2022-06-23:
This is not the only problem with 1.0.0 (see the related discussion), so I changed the bug synopsis.

prusnak, 2022-10-18:
Possible fixes in https://github.com/trustcrypto/onlykey-solo-python/commit/7e570dc0f7292eded4644f4eb637e22308aebd91 (files `solo/devices/base.py`, `solo/devices/solo_v1.py`, `solo/solotool.py`)

mcepl, 2023-07-29:
> Possible fixes in [trustcrypto@7e570dc](https://github.com/trustcrypto/onlykey-solo-python/commit/7e570dc0f7292eded4644f4eb637e22308aebd91) (files `solo/devices/base.py`, `solo/devices/solo_v1.py`, `solo/solotool.py`)

There are not, this is just a subset of #159 and pretty poorly done (removing some files and adding some others without a reason).

innir, 2023-10-14:
@nickray @kms15 any update on the fido2 >= 1.0 transition. Debian would like to get fido2 1.1 in and solo1-cli is currently blocking.

innir, 2023-10-17:
I started working on this not in PR #169 - it's very much WIP. Please test & report breakages as I don't really have an idea how to systematically test ...

innir, 2023-12-07:
@nickray If you don't act now, this will kick solo1-cli out of Debian, Ubuntu and all their downstream distros unfortunately.

fabaff, 2024-05-24:
> ...will kick solo1-cli out of Debian, Ubuntu and all their downstream distros unfortunately.

The same applies for NIxpkgs.

innir, 2024-05-24:
@fabaff You could use PR #169. This version is in Debian testing for six months now without bug reports.