solokeys/solo1-cli #174: [Question] Does not work with fido2 >= 1.0.0 ?

aanno, 2024-07-14:
Can use and/or build with fido2 >= 1.0.0.

```bash
make
...
ImportError: cannot import name 'CTAP1' from 'fido2.ctap1' (/stratis/home/tpasch/dev/scm/github/solo1-cli/venv/lib/python3.11/site-packages/fido2/ctap1.py)
make: *** [Makefile:54: update-venv] Fehler 1
```

Same for using package from pypi (`pip3 install --user solo1`)

Work around:

```bash
python3 -m venv .venv
. .venv/bin/activate
pip3 install --force-reinstall -v "fido2==0.9.3" solo1
```

Also, see https://github.com/Nike-Inc/gimme-aws-creds/issues/355

aanno, 2024-07-14:
Ah, https://github.com/solokeys/solo1-cli/pull/169 mentions the PR https://github.com/trustcrypto/onlykey-solo-python/pull/3/files that is still a _draft_.

Also see https://github.com/solokeys/solo1-cli/issues/172

troysandal, 2024-11-20:
A thumbs to the draft PR above.  I went the opposite direction and got things working by pinning to `fido2 < 1.0.0`.

See https://pastebin.com/FLdyy8dE